tenstorrent/tt-flash #1: Build and Publish wheels for each release

hmohiuddinTT, 2023-12-07:
To avoid users having to build wheels themselves we should release the wheels on GitHub Releases along with the src code.

We can setup GitHub actions to trigger on release which can generate the wheels and upload to the same release.

The same workflow can also be used to push the package to PyPi once ready.

warthog9, 2024-04-26:
Just so this isn't hidden way, this is still in flight, mostly I want to get distro packages (rpm and deb) sorted at the same time here as I don't want to have the wheels be the only method out there and folks get used to them being the only way (mostly because upgrading that way isn't as convenient).

I've got luwen building rpms privately right now, and debs are most of the way-ish there.  Just so there's a noted update here.

hmohiuddinTT, 2024-05-06:
Hey @warthog9, wondering if there's any updates on this. Cloud is blocked from upgrading to the latest firmware on our machines because we do not want to be building from source. 

Is it possible to increase the priority on this?

warthog9, 2024-10-30:
progress but slow for now just as an update

warthog9, 2025-10-15:
out on pypi

closed 2025-10-15 by warthog9