tenstorrent/tt-flash #3: Unable to install tt-flash

eyonland, 2024-01-15:
Hitting an error following instructions from [ https://docs.tenstorrent.com/tenstorrent/add-in-boards-and-cooling-kits/e75-e150/software-setup#install-tt-flash]( https://docs.tenstorrent.com/tenstorrent/add-in-boards-and-cooling-kits/e75-e150/software-setup#install-tt-flash)

pip install https://github.com/tenstorrent/tt-flash.git
Defaulting to user installation because normal site-packages is not writeable
Collecting https://github.com/tenstorrent/tt-flash.git
  Downloading https://github.com/tenstorrent/tt-flash.git
     \ 186.4 kB 2.1 MB/s 0:00:00
  ERROR: Cannot unpack file /tmp/pip-unpack-hk668wcf/tt-flash.git (downloaded from /tmp/pip-req-build-9ra7jm6b, content-type: text/html; charset=utf-8); cannot detect archive format
ERROR: Cannot determine archive format of /tmp/pip-req-build-9ra7jm6b

eyonland, 2024-01-15:
python3 --version
Python 3.10.12

pip --version
pip 22.0.2 from /usr/lib/python3/dist-packages/pip (python 3.10)

TTDRosen, 2024-01-16:
It looks like the install instructions on the docs website are incorrect.

Try `pip install git+https://github.com/tenstorrent/tt-flash.git` and let me know if you get better results

eyonland, 2024-01-16:
Yep that worked beautifully!

TTDRosen, 2024-01-16:
Great! Thanks for filing the issue 😄 I'll get the docs updated to stop others form hitting this same problem.

eyonland, 2024-01-16:
It looks like tt-flash did not get setup in the $HOME/.local/bin/tt-flash location though.

Using a newer version on Ubuntu by the way...
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

TTDRosen, 2024-01-16:
Try running `pip list -v` that'll show you if/where pip installed tt-flash

eyonland, 2024-01-16:
tt-flash does not show up.

eyonland, 2024-01-16:
I was able to get tt-flash setup properly after the following steps:

1. Installing cargo with `sudo apt install cargo`.  
2. Used pyenv to target version 3.8.10.  
3. pip install git+https://github.com/tenstorrent/tt-flash.git

warthog9, 2024-02-05:
Docs fixed

closed 2024-02-05 by warthog9