tenstorrent/tt-flash #6: Trap old firmware traceback slightly better

warthog9, 2024-02-03:
```
(venv) root@orko:~/git/tt-firmware-gs# tt-flash --fw-tar fw_pack-80.4.0.0_acec1267.tar.gz

Now checking device <builtins.PciChip object at 0x7f94983a21c0>:

Traceback (most recent call last):
  File "/root/venv/bin/tt-flash", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/root/venv/lib64/python3.12/site-packages/tt_flash/main.py", line 458, in main
    handle_args(chip, tar, manifest.get("bundle_version", {}), args)
  File "/root/venv/lib64/python3.12/site-packages/tt_flash/main.py", line 227, in handle_args
    raise TTError("Looks like you are running a very old set of fw, it's safe to assume that it needs an update but please update it using --force")
tt_flash.error.TTError: Looks like you are running a very old set of fw, it's safe to assume that it needs an update but please update it using --force
(venv) root@orko:~/git/tt-firmware-gs#
```

It's good that it gets thrown and it spits out the error but it would be better if we suppressed the traceback and just kicked out the error, maybe colorized would be nice(?).  The traceback just looks like the program failed (well it did technically), but in a way that it's an issue that isn't solvable per-se

afaerber, 2024-09-10:
Negative, still a traceback in 3.0.0:
```
sudo tt-flash flash --fw-tar fw_pack-80.10.0.0.fwbundle 
Stage: SETUP
	Searching for default sys-config path
	Checking /etc/tenstorrent/config.json: not found
	Checking ~/.config/tenstorrent/config.json: not found

	Could not find config in default search locations, if you need it, either pass it in explicity or generate one
	Warning: continuing without sys-config, galaxy systems will not be reset
Stage: DETECT
Stage: FLASH
	Sub Stage: VERIFY
		Verifying fw-package can be flashed: complete
		Verifying Wormhole[0] can be flashed
	Stage: FLASH
		Sub Stage FLASH Step 1: Wormhole[0]
Traceback (most recent call last):
  File "/usr/bin/tt-flash", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.11/site-packages/tt_flash/main.py", line 249, in main
    return flash_chips(
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/tt_flash/flash.py", line 677, in flash_chips
    result = flash_chip_stage1(
             ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/tt_flash/flash.py", line 232, in flash_chip_stage1
    raise TTError(
tt_flash.error.TTError: Looks like you are running a very old set of fw, it's safe to assume that it needs an update but please update it using --force
```

TTDRosen, 2024-09-10:
It's still on my list, and will likely be fixed in or around 3.2.0. Currently adding firmware verification and logging to syslog are ranking above usability improvements.

closed 2025-10-15 by warthog9