tenstorrent/tt-flash #8: package_root_path() in tt-flash/utility.py

frkl, 2024-03-15:
I was getting some errors when running from Python 3.9
```
(tt) grrrgrrr@grrrgrrr-linuxbox3:~$ tt-flash
Traceback (most recent call last):
  File "/home/grrrgrrr/anaconda3/envs/tt/bin/tt-flash", line 5, in <module>
    from tt_flash.main import main
  File "/home/grrrgrrr/anaconda3/envs/tt/lib/python3.9/site-packages/tt_flash/main.py", line 22, in <module>
    with utility.package_root_path() as path:
  File "/home/grrrgrrr/anaconda3/envs/tt/lib/python3.9/contextlib.py", line 119, in __enter__
    return next(self.gen)
  File "/home/grrrgrrr/anaconda3/envs/tt/lib/python3.9/importlib/resources.py", line 175, in _path_from_reader
    opener_reader = reader.open_resource(norm_resource)
  File "<frozen importlib._bootstrap_external>", line 1055, in open_resource
IsADirectoryError: [Errno 21] Is a directory: '/home/grrrgrrr/anaconda3/envs/tt/lib/python3.9/site-packages/tt_flash'

```
After some digging, it seems that this error has to do with `package_root_path()` in `tt-flash/utility.py`. Current implementation uses `importlib.resources.path`
```
# Returns the root path of the package, so we can access data files and such
def package_root_path():
    return path("tt_flash", "")
```
It seems to have been dropped in favor of `importlib.resources.files()` per [https://stackoverflow.com/a/77058615](https://stackoverflow.com/a/77058615)

Here's a proposed change that seems to work for Python 3.9.

```
from importlib.resources import files
# Returns the root path of the package, so we can access data files and such
def package_root_path():
    return files("tt_flash")
```

warthog9, 2024-04-25:
Ok quick testing, python3.8 works fine, python3.9 doesn't, python3.12 does - so there's been a clear patch in the core library going backwards somewhere between 3.9 and 3.12.

@TTDRosen when you get a chance, can you chase down what's going on there and figure out what the correct way to do that is?

TTDRosen, 2024-05-08:
Finally got a chance to resolve this, as of v2.0.7 Python3.9 support is no longer broken.

closed 2024-05-08 by TTDRosen