tenstorrent/tt-flash #9: Missing --fw-tar option from --help menu.

hmohiuddinTT, 2024-03-21:
Missing the --fw-tar option from the --help menu.

[image]

Related to: https://github.com/tenstorrent/tt-flash/blob/9808227dc84ec5e0d5a81d8860b04f9e31338159/tt_flash/main.py#L70C1-L71C1

warthog9, 2024-04-25:
That's odd it looks like it should be there already:

https://github.com/tenstorrent/tt-flash/blame/bd53bc043189baa8324b8ffbb9454ee3650c54b6/tt_flash/main.py#L70

When you get a chance can you re-test with latest?

warthog9, 2024-04-25:
> (venv) root@orko:~# tt-flash --help
> usage: tt-flash [-h] [-v] [--force] [--fw-tar FW_TAR]
> 
> options:
>   -h, --help       show this help message and exit
>   -v, --version    show program's version number and exit
>   --force          Force update the ROM
>   --fw-tar FW_TAR  Path to the firmware tarball
> (venv) root@orko:~#

I think this is already resolved so going to close this

warthog9, 2024-04-25:
Ok re-opening as I figured out what's going on, and this is valid.  The help text *EXISTS*, except when you run the top level `tt-flash --help` or `tt-flash` without anything it shows the options for `tt-flash` but *DOES NOT* show that the commands exist.  So we need a way to get what commands are available into the top level help more obviously

afaerber, 2024-09-10:
As visible at the bottom of the first screenshot, it does show the commands:
```
command:
  {flash}
```

What's missing is a sentence to the effect of, "See tt-flash {command} --help for further command-specific help."