tenstorrent/tt-flash #20: Error during installing tt-flash at "Preparing Metadata"

saichandax, 2024-12-26:
I encountered the following error while installing tt-flash on n300.

```
(.venv) mcw@mcwTT:~$ git clone https://github.com/tenstorrent/tt-flash.git
Cloning into 'tt-flash'...
remote: Enumerating objects: 322, done.
remote: Counting objects: 100% (78/78), done.
remote: Compressing objects: 100% (30/30), done.
remote: Total 322 (delta 65), reused 48 (delta 48), pack-reused 244 (from 1)
Receiving objects: 100% (322/322), 105.60 KiB | 2.58 MiB/s, done.
Resolving deltas: 100% (193/193), done.
(.venv) mcw@mcwTT:~$ cd tt-flash
(.venv) mcw@mcwTT:~/tt-flash$ pip install .
Looking in indexes: https://pypi.org/simple, https://download.pytorch.org/whl/cpu
Processing /home/mcw/tt-flash
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pyluwen@ git+https://github.com/tenstorrent/luwen.git@v0.4.6#subdirectory=crates/pyluwen (from tt-flash==3.1.0)
  Cloning https://github.com/tenstorrent/luwen.git (to revision v0.4.6) to /tmp/pip-install-rclcb706/pyluwen_a2550f41c4f74e1b87db80e3b7d2539a
  Running command git clone --filter=blob:none --quiet https://github.com/tenstorrent/luwen.git /tmp/pip-install-rclcb706/pyluwen_a2550f41c4f74e1b87db80e3b7d2539a
  Running command git checkout -q 5cf4cc6545b7d9d92f167417faf39822ed88dbf7
  Resolved https://github.com/tenstorrent/luwen.git to commit 5cf4cc6545b7d9d92f167417faf39822ed88dbf7
  Running command git submodule update --init --recursive -q
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [12 lines of output]
          Updating crates.io index
           Locking 191 packages to latest compatible versions
            Adding bitfield-struct v0.6.2 (available: v0.10.0)
            Adding memmap2 v0.7.1 (available: v0.9.5)
            Adding nix v0.26.4 (available: v0.29.0)
            Adding pyo3 v0.19.2 (available: v0.23.3)
            Adding thiserror v1.0.69 (available: v2.0.9)
      💥 maturin failed
        Caused by: `project.version` field is required in pyproject.toml unless it is present in the `project.dynamic` list
      Error running maturin: Command '['maturin', 'pep517', 'write-dist-info', '--metadata-directory', '/tmp/pip-modern-metadata-1oimamkk', '--interpreter', '/home/mcw/.venv/bin/python3']' returned non-zero exit status 1.
      Checking for Rust toolchain....
      Running `maturin pep517 write-dist-info --metadata-directory /tmp/pip-modern-metadata-1oimamkk --interpreter /home/mcw/.venv/bin/python3`
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
(.venv) mcw@mcwTT:~/tt-flash$ 
``` 

Device: n300
OS: ubuntu-20.04

Please let me know if I missed something.

goshlanguage, 2024-12-27:
I can replicate this on Ubuntu-22, I suspect it's the `maturin` dependency and how it's defined. I can reproduce with the following:

```sh
git clone https://github.com/tenstorrent/tt-flash
cd tt-flash

sudo apt install python3.10-venv

python3 -m venv venv
source venv/bin/activate

pip install .
```

I can also reproduce using `pip install git+https://github.com/tenstorrent/tt-flash.git` as referenced in the `README.md`

windwardly, 2024-12-30:
Looks like the fix is started in [luwen](https://github.com/tenstorrent/luwen/commit/754b8b6d09acb4aacbee03465e040124ea420b75)

While waiting for how Tenstorrent wants to fix it, you can change this in pyproject.toml to drop the version and just use tip:

'
"pyluwen @ git+https://github.com/tenstorrent/luwen.git@v0.4.6#subdirectory=crates/pyluwen",

to

"pyluwen @ git+https://github.com/tenstorrent/luwen.git#subdirectory=crates/pyluwen",
`

While including other changes, it seems to work.

milank94, 2025-01-03:
Issue has been resolved as of: https://github.com/tenstorrent/luwen/pull/12

Please try to install again.

windwardly, 2025-01-03:
My sense is that the fix was related to further work in `maturin` with [2417](https://github.com/PyO3/maturin/pull/2417) and [2418](https://github.com/PyO3/maturin/pull/2418), which shifted the error to  be a warning except in `maturin develop`.

The `luwen` that's linked into `tt-flash` (and `tt-smi`, `tt-topology` and maybe others) is at a fixed version in `pyproject.toml`, so looks like the new change to `luwen` is not brought in:
```
"pyluwen @ git+https://github.com/tenstorrent/luwen.git@v0.4.6#subdirectory=crates/pyluwen"
```

That's just details however -- the bottom line matches: `tt-flash` works again!

warthog9, 2025-01-06:
tt-smi was pushed as a tag but apparently missed the main branch, I missed flash apparently and topology had been fixed but the push was messed up slightly - sorted now.  Sorry about that, I'm going to blame the holiday

windwardly, 2025-01-06:
All looks fine. Yes, me too blaming the holidays, moving quickly between different branches on repos, I likely missed the tt-smi update.

Thank you.

warthog9, 2025-01-06:
@saichandax can you confirm this is closeable at this point?

saichandax, 2025-01-07:
We can close this. Thanks for the fix. 👍

closed 2025-01-07 by milank94