tenstorrent/tt-flash #48: Flashing firmware causes bus issues on Ampere Altra

RossComputerGuy, 2025-08-14:
After bumping my Blackhole p150a from 18.0.0 to 18.8.0, I get the following bus issue:
```
thread '<unnamed>' panicked at crates/luwen-if/src/chip/communication/chip_comms.rs:58:37:
called `Option::unwrap()` on a `None` value
stack backtrace:
   0:     0xfffd9dd8c24c - <std::sys::backtrace::BacktraceLock::print::DisplayBacktrace as core::fmt::Display>::fmt::ha1e40045faa5a504
   1:     0xfffd9ddaa1bc - core::fmt::write::hdf82382eb63ea667
   2:     0xfffd9dd7f458 - std::io::Write::write_fmt::hba26b7a93568e182
   3:     0xfffd9dd8d024 - std::panicking::default_hook::{{closure}}::hb4699d15d1a38e40
   4:     0xfffd9dd8cef0 - std::panicking::default_hook::h238c6097846d6da4
   5:     0xfffd9dd8d688 - std::panicking::rust_panic_with_hook::h2b5d48eaef142e4c
   6:     0xfffd9dd8d3f0 - std::panicking::begin_panic_handler::{{closure}}::h2edd966f50ed08e4
   7:     0xfffd9dd8c48c - std::sys::backtrace::__rust_end_short_backtrace::hc15bc42cd3c6c7df
   8:     0xfffd9dd8d0c8 - __rustc[b723cc1e8dfe1f7e]::rust_begin_unwind
   9:     0xfffd9db8c67c - core::panicking::panic_fmt::h7b3dfc9925559f98
  10:     0xfffd9db8c6ec - core::panicking::panic::h2019ec3d2a38bf44
  11:     0xfffd9db8ca18 - core::option::unwrap_failed::h30dd2decc9eae9a5
  12:     0xfffd9dcd89b8 - core::option::Option<T>::unwrap::h7e72f3f9bfa1a85e
                               at /build/rustc-1.88.0-src/library/core/src/option.rs:1003:21
  13:     0xfffd9dcd89b8 - luwen_if::chip::communication::chip_comms::load_axi_table::h4f47daeba5042024
                               at /build/source/crates/luwen-if/src/chip/communication/chip_comms.rs:58:16
  14:     0xfffd9dc4e370 - luwen_if::chip::creation::<impl luwen_if::chip::Chip>::bh_open::h5ad579ec87e50576
                               at /build/source/crates/luwen-if/src/chip/creation.rs:85:27
  15:     0xfffd9dc4e018 - luwen_if::chip::creation::<impl luwen_if::chip::Chip>::open::h731ebef6962a66bb
                               at /build/source/crates/luwen-if/src/chip/creation.rs:117:45
  16:     0xfffd9db99d44 - pyluwen::PciChip::new::h3e22f41060fa18d0
                               at /build/source/crates/pyluwen/src/lib.rs:657:13
  17:     0xfffd9dbbcb44 - pyluwen::detect_chips_fallible::h38b2b0cb2c88dd3c
                               at /build/source/crates/pyluwen/src/lib.rs:1609:20
  18:     0xfffd9dbbdfa0 - pyluwen::_::__pyfunction_detect_chips_fallible::h59547bf16f6e2431
                               at /build/source/crates/pyluwen/src/lib.rs:1574:1
  19:     0xfffd9dbe0298 - pyo3::impl_::trampoline::fastcall_with_keywords::{{closure}}::h3d1824e20674428d
                               at /build/cargo-deps-vendor/pyo3-0.19.2/src/impl_/trampoline.rs:41:29
  20:     0xfffd9dbdfedc - pyo3::impl_::trampoline::trampoline::{{closure}}::h424079de18977330
                               at /build/cargo-deps-vendor/pyo3-0.19.2/src/impl_/trampoline.rs:181:54
  21:     0xfffd9dbcf994 - std::panicking::try::do_call::h38d635e0b885926d
                               at /build/rustc-1.88.0-src/library/std/src/panicking.rs:589:40
  22:     0xfffd9dbe5bb4 - __rust_try
  23:     0xfffd9dbe5b18 - std::panicking::try::h35fa2a4f951c139d
                               at /build/rustc-1.88.0-src/library/std/src/panicking.rs:552:19
  24:     0xfffd9dbe5b18 - std::panic::catch_unwind::he0cd33fdaf735f34
                               at /build/rustc-1.88.0-src/library/std/src/panic.rs:359:14
  25:     0xfffd9dbdf9ec - pyo3::impl_::trampoline::trampoline::h30d407cc60a2e6e0
                               at /build/cargo-deps-vendor/pyo3-0.19.2/src/impl_/trampoline.rs:181:9
  26:     0xfffd9dbe8740 - pyo3::impl_::trampoline::fastcall_with_keywords::hd8eea6884444d918
                               at /build/cargo-deps-vendor/pyo3-0.19.2/src/impl_/trampoline.rs:41:13
  27:     0xfffd9dbbda50 - pyluwen::_::<impl pyluwen::detect_chips_fallible::MakeDef>::DEF::trampoline::h80c77e845a76d5e0
                               at /build/source/crates/pyluwen/src/lib.rs:1574:1
  28:     0xfffda01246a4 - cfunction_vectorcall_FASTCALL_KEYWORDS
  29:     0xfffda012dad4 - PyObject_Vectorcall.localalias
  30:     0xfffda035f518 - _PyEval_EvalFrameDefault.localalias
  31:     0xfffda036eab0 - PyEval_EvalCode.localalias
  32:     0xfffda0374258 - run_eval_code_obj
  33:     0xfffda03745d8 - run_mod
  34:     0xfffda0376c0c - _PyRun_SimpleFileObject
  35:     0xfffda0378dc4 - Py_RunMain.localalias
  36:     0xfffd9fcd62b4 - __libc_start_call_main
  37:     0xfffd9fcd6398 - __libc_start_main@@GLIBC_2.34
  38:           0x400970 - _start
  39:                0x0 - <unknown>
Traceback (most recent call last):
  File "/nix/store/3shi2qm6w5ssp14fbl66x46blziv0hx6-tt-smi-3.0.21/bin/.tt-smi-wrapped", line 9, in <module>
    sys.exit(main())
             ~~~~^^
  File "/nix/store/3shi2qm6w5ssp14fbl66x46blziv0hx6-tt-smi-3.0.21/lib/python3.13/site-packages/tt_smi/tt_smi.py", line 923, in main
    devices = detect_chips_with_callback(
        local_only=args.local, ignore_ethernet=args.local, print_status=is_tty
    )
  File "/nix/store/0gh2y4bg7837s66f647j49b1wm9i9n91-tt-tools-common-1.4.16/lib/python3.13/site-packages/tt_tools_common/utils_common/tools_utils.py", line 321, in detect_chips_with_callback
    for device in detect_chips_fallible(
                  ~~~~~~~~~~~~~~~~~~~~~^
        local_only=local_only,
        ^^^^^^^^^^^^^^^^^^^^^^
    ...<2 lines>...
        noc_safe=ignore_ethernet,
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    ):
    ^
pyo3_runtime.PanicException: called `Option::unwrap()` on a `None` value
```

RossComputerGuy, 2025-08-17:
That's a good question, the thing is tt-smi worked before the firmware update but I wonder if updating caused something to change in luwen if luwen does version checks.

milank94, 2025-09-02:
@RossComputerGuy were you able to try a fresh environment install?